airalab/robonomics_comm #9: Gas limit on workers

EvgenRad81, 2018-07-23:
It seems, the gas limit is incorrectly calculated on workers.

https://etherscan.io/tx/0xc2ae1e7567db1840ae0cc8c42648c5e6d3979ed10dd3dcd7c22aaf10e04dce07
https://etherscan.io/tx/0xc3238e2e5d01ea16cf69448663d2492e13afb3ec83804423e72bb73eb3414e1c

I propose to make a calculated gas +100k

akru, 2018-07-23:
In my opinion if gas estimator lie then +100k gas is not a solution because it doesn't work at all time. 

My proposal is static limit for transactions: 950k gas for liability creation and 300k gas for liability finalization.

EvgenRad81, 2018-07-24:
I think +100к is more universal. Сontracts may change.  You need to keep one more thing in your head.))

akru, 2018-07-27:
No, it'll not work, as you can see here https://etherscan.io/tx/0x3dd30f86a42fdbc1ea3886531fc684a455fa9ae04de0ca1a22d62fc705387768 for liability finalization 200k gas is required, but in your txs available 230k gas, this is mean that fail is cathed from `throw` function call.

closed 2018-07-27 by akru