airalab/robonomics_comm #15: [rospy.internal][ERROR] : Inbound TCP/IP connection failed

khssnv, 2018-07-30:
Sometimes the provider (lighthouse service) fails with an error shown below. I can't figure out circumstances, sometimes it may work a day or longer, sometimes no more than an hour.
```bash
[rosout][INFO] 2018-07-30 12:59:51,962: Transaction added to queue
[rosout][INFO] 2018-07-30 12:59:52,175: Transaction GAS 902324, price 10 gwei
[rosout][INFO] 2018-07-30 12:59:52,982: Transaction sended at 0x015879b34815ca691c147eed5a8312595fe7ff6e4796bd18ee6bdbcf3caceb9d
[rosout][INFO] 2018-07-30 13:01:53,349: Transaction mined at 0x015879b34815ca691c147eed5a8312595fe7ff6e4796bd18ee6bdbcf3caceb9d
[rosout][INFO] 2018-07-30 13:01:53,826: Lighthouse m: 1 q: 1 k: 6057362, b: 6057344
[rosout][WARNING] 2018-07-30 13:05:29,029: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[rospy.internal][ERROR] 2018-07-30 13:05:29,032: Inbound TCP/IP connection failed:
Traceback (most recent call last):
  File "/nix/store/qx7wqz3aaw318lcsrs3p66p81lh2dyg5-rospy-1.13.6/lib/python3.6/site-packages/rospy/impl/tcpros_base.py", line 324, in _tcp_server_callback
    header = read_ros_handshake_header(sock, BytesIO(), buff_size)
  File "/nix/store/hppzwfkaf2xg6nnrwldlc5332ky6r67n-rosgraph-1.13.6/lib/python3.6/site-packages/rosgraph/network.py", line 364, in read_ros_handshake_header
    raise ROSHandshakeException("connection from sender terminated before handshake header received. %s bytes were received. Please check sender for additional details."%b.tell())
rosgraph.network.ROSHandshakeException: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.

[rosout][WARNING] 2018-07-30 13:05:49,185: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 229 bytes were received. Please check sender for additional details.
[rospy.internal][ERROR] 2018-07-30 13:05:49,186: Inbound TCP/IP connection failed:
Traceback (most recent call last):
  File "/nix/store/qx7wqz3aaw318lcsrs3p66p81lh2dyg5-rospy-1.13.6/lib/python3.6/site-packages/rospy/impl/tcpros_base.py", line 324, in _tcp_server_callback
    header = read_ros_handshake_header(sock, BytesIO(), buff_size)
  File "/nix/store/hppzwfkaf2xg6nnrwldlc5332ky6r67n-rosgraph-1.13.6/lib/python3.6/site-packages/rosgraph/network.py", line 364, in read_ros_handshake_header
    raise ROSHandshakeException("connection from sender terminated before handshake header received. %s bytes were received. Please check sender for additional details."%b.tell())
rosgraph.network.ROSHandshakeException: connection from sender terminated before handshake header received. 229 bytes were received. Please check sender for additional details.

[rosout][WARNING] 2018-07-30 13:06:09,342: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 45 bytes were received. Please check sender for additional details.
[rospy.internal][ERROR] 2018-07-30 13:06:09,343: Inbound TCP/IP connection failed:
Traceback (most recent call last):
  File "/nix/store/qx7wqz3aaw318lcsrs3p66p81lh2dyg5-rospy-1.13.6/lib/python3.6/site-packages/rospy/impl/tcpros_base.py", line 324, in _tcp_server_callback
    header = read_ros_handshake_header(sock, BytesIO(), buff_size)
  File "/nix/store/hppzwfkaf2xg6nnrwldlc5332ky6r67n-rosgraph-1.13.6/lib/python3.6/site-packages/rosgraph/network.py", line 364, in read_ros_handshake_header
    raise ROSHandshakeException("connection from sender terminated before handshake header received. %s bytes were received. Please check sender for additional details."%b.tell())
rosgraph.network.ROSHandshakeException: connection from sender terminated before handshake header received. 45 bytes were received. Please check sender for additional details.
```

khssnv, 2018-07-30:
I just find out that lighthouse node continuous to work after error messages. Please find full log below.

<details>
  <summary>lighthouse-lighthouse-6.log</summary>

```bash
[rospy.client][INFO] 2018-07-30 14:31:06,631: init_node, name[/lighthouse/lighthouse], pid[32437]
[xmlrpc][INFO] 2018-07-30 14:31:06,631: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2018-07-30 14:31:06,632: Started XML-RPC server [http://promisor-node:36509/]
[rospy.impl.masterslave][INFO] 2018-07-30 14:31:06,634: _ready: http://promisor-node:36509/
[rospy.registration][INFO] 2018-07-30 14:31:06,635: Registering with master node http://localhost:11311
[rospy.init][INFO] 2018-07-30 14:31:06,640: ROS Slave URI: [http://promisor-node:36509/]
[xmlrpc][INFO] 2018-07-30 14:31:06,640: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2018-07-30 14:31:06,640: registered with master
[rospy.rosout][INFO] 2018-07-30 14:31:06,641: initializing /rosout core topic
[rospy.rosout][INFO] 2018-07-30 14:31:06,646: connected to core topic /rosout
[rospy.simtime][INFO] 2018-07-30 14:31:06,648: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2018-07-30 14:31:06,893: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2018-07-30 14:31:06,912: topic[/lighthouse/deal] adding connection to [http://promisor-node:37093/], count 0
[rospy.internal][INFO] 2018-07-30 14:31:06,926: topic[/lighthouse/infochan/incoming/result] adding connection to [http://promisor-node:37947/], count 0
[rosout][INFO] 2018-07-30 14:31:07,552: Lighthouse m: 1 q: 1 k: 6057362, b: 6057709
[rosout][WARNING] 2018-07-30 14:45:30,494: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[rospy.internal][ERROR] 2018-07-30 14:45:30,747: Inbound TCP/IP connection failed:
Traceback (most recent call last):
  File "/nix/store/qx7wqz3aaw318lcsrs3p66p81lh2dyg5-rospy-1.13.6/lib/python3.6/site-packages/rospy/impl/tcpros_base.py", line 324, in _tcp_server_callback
    header = read_ros_handshake_header(sock, BytesIO(), buff_size)
  File "/nix/store/hppzwfkaf2xg6nnrwldlc5332ky6r67n-rosgraph-1.13.6/lib/python3.6/site-packages/rosgraph/network.py", line 364, in read_ros_handshake_header
    raise ROSHandshakeException("connection from sender terminated before handshake header received. %s bytes were received. Please check sender for additional details."%b.tell())
rosgraph.network.ROSHandshakeException: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.

[rosout][WARNING] 2018-07-30 14:45:50,549: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 229 bytes were received. Please check sender for additional details.
[rospy.internal][ERROR] 2018-07-30 14:45:50,550: Inbound TCP/IP connection failed:
Traceback (most recent call last):
  File "/nix/store/qx7wqz3aaw318lcsrs3p66p81lh2dyg5-rospy-1.13.6/lib/python3.6/site-packages/rospy/impl/tcpros_base.py", line 324, in _tcp_server_callback
    header = read_ros_handshake_header(sock, BytesIO(), buff_size)
  File "/nix/store/hppzwfkaf2xg6nnrwldlc5332ky6r67n-rosgraph-1.13.6/lib/python3.6/site-packages/rosgraph/network.py", line 364, in read_ros_handshake_header
    raise ROSHandshakeException("connection from sender terminated before handshake header received. %s bytes were received. Please check sender for additional details."%b.tell())
rosgraph.network.ROSHandshakeException: connection from sender terminated before handshake header received. 229 bytes were received. Please check sender for additional details.

[rosout][WARNING] 2018-07-30 14:46:10,603: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 45 bytes were received. Please check sender for additional details.
[rospy.internal][ERROR] 2018-07-30 14:46:10,604: Inbound TCP/IP connection failed:
Traceback (most recent call last):
  File "/nix/store/qx7wqz3aaw318lcsrs3p66p81lh2dyg5-rospy-1.13.6/lib/python3.6/site-packages/rospy/impl/tcpros_base.py", line 324, in _tcp_server_callback
    header = read_ros_handshake_header(sock, BytesIO(), buff_size)
  File "/nix/store/hppzwfkaf2xg6nnrwldlc5332ky6r67n-rosgraph-1.13.6/lib/python3.6/site-packages/rosgraph/network.py", line 364, in read_ros_handshake_header
    raise ROSHandshakeException("connection from sender terminated before handshake header received. %s bytes were received. Please check sender for additional details."%b.tell())
rosgraph.network.ROSHandshakeException: connection from sender terminated before handshake header received. 45 bytes were received. Please check sender for additional details.

[rosout][WARNING] 2018-07-30 14:47:16,086: Inbound TCP/IP connection failed: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[rospy.internal][ERROR] 2018-07-30 14:47:16,087: Inbound TCP/IP connection failed:
Traceback (most recent call last):
  File "/nix/store/qx7wqz3aaw318lcsrs3p66p81lh2dyg5-rospy-1.13.6/lib/python3.6/site-packages/rospy/impl/tcpros_base.py", line 324, in _tcp_server_callback
    header = read_ros_handshake_header(sock, BytesIO(), buff_size)
[... cut ...]
[rosout][INFO] 2018-07-30 15:17:11,690: Transaction added to queue
[rosout][INFO] 2018-07-30 15:17:11,907: Transaction GAS 860656, price 10 gwei
[rosout][ERROR] 2018-07-30 15:17:21,997: Broken transaction: HTTPConnectionPool(host='localhost', port=8545): Read timed out. (read timeout=10)
[rosout][INFO] 2018-07-30 15:17:22,502: Lighthouse m: 1 q: 1 k: 6057362, b: 6057899
[rosout][INFO] 2018-07-30 15:36:51,539: Transaction added to queue
[rosout][INFO] 2018-07-30 15:36:51,731: Transaction GAS 903834, price 10 gwei
```
</details>

akru, 2018-07-31:
This seems as `rospy` bug, need more info to reproduce it locally.

khssnv, 2018-07-31:
Another error message "Inbound TCP/IP connection" which not stop a node, but looks weird. Please find it below. Do you think we should report to rospy devs?

<details>
  <summary>lighthouse-lighthouse-6.log</summary>

```bash
[rosout][INFO] 2018-07-31 13:33:50,970: Transaction added to queue
[rosout][INFO] 2018-07-31 13:34:20,977: Transaction added to queue
[rosout][INFO] 2018-07-31 13:34:50,986: Transaction added to queue
[rosout][WARNING] 2018-07-31 13:35:02,147: Inbound TCP/IP connection failed: [Errno 104] Connection reset by peer
[rospy.internal][ERROR] 2018-07-31 13:35:02,150: Inbound TCP/IP connection failed:
Traceback (most recent call last):
  File "/nix/store/qx7wqz3aaw318lcsrs3p66p81lh2dyg5-rospy-1.13.6/lib/python3.6/site-packages/rospy/impl/tcpros_base.py", line 324, in _tcp_server_callback
    header = read_ros_handshake_header(sock, BytesIO(), buff_size)
  File "/nix/store/hppzwfkaf2xg6nnrwldlc5332ky6r67n-rosgraph-1.13.6/lib/python3.6/site-packages/rosgraph/network.py", line 362, in read_ros_handshake_header
    d = sock.recv(buff_size)
ConnectionResetError: [Errno 104] Connection reset by peer

[rosout][INFO] 2018-07-31 13:35:20,989: Transaction added to queue
[rosout][INFO] 2018-07-31 13:35:51,029: Transaction added to queue
[rosout][INFO] 2018-07-31 13:36:21,025: Transaction added to queue
```
</details>

akru, 2018-07-31:
Yes, it could be a reason to ask about rostcp behavior.