airalab/robonomics_comm #19: More web3 info from ethereum_common

khssnv, 2018-08-16:
Hi!

I need a few more web3 things from the ethereum_common node.
* _current block number_ - to let lot publisher know where to start while computing deadline,
* _eth / token balance_ - to let CPS know its income,
* _approved amount_ - to check how much tokens already approved to liability factory and should I approve more for next operation.

Can you implement it or give a direction on how to do it? Seems like ERC20 node is not a place for things like balance or block number.

akru, 2018-08-21:
@gitcoinbot start work

gitcoinbot, 2018-08-21:
Issue Status: 1. **Open** 2. Started 3. Submitted 4. Done 

<hr>

__This issue now has a funding of 0.1 ETH (27.76 USD @ $277.56/ETH)  attached to it.__

 * If you would like to work on this issue you can 'start work' [on the Gitcoin Issue Details page](https://gitcoin.co/issue/airalab/robonomics_comm/19/999).
* Want to chip in? Add your own contribution [here](https://gitcoin.co/issue/airalab/robonomics_comm/19/999).
* Questions? Checkout <a href='https://gitcoin.co/help'>Gitcoin Help</a> or the <a href='https://gitcoin.co/slack'>Gitcoin Slack</a>
* $18,964.73 more funded OSS Work available on the [Gitcoin Issue Explorer](https://gitcoin.co/explorer)

gitcoinbot, 2018-08-23:
Issue Status: 1. **Open** 2. Started 3. Submitted 4. Done 

<hr>

__Workers have applied to start work__.

These users each claimed they can complete the work by 4 weeks from now.
Please review their action plans below:

1. [Rohithzr](https://gitcoin.co/profile/Rohithzr) has applied to start work _(Funders only: [approve worker](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?mutate_worker_action=approve&worker=Rohithzr) | [reject worker](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?mutate_worker_action=reject&worker=Rohithzr))_ (1st dibs). 
    
    so I have been working on web3js and web3py since last year and have seen it grow to what it is today lets take your issues for example.

current block number - there are many ways to know the block number from hash, address, contract address or just your node's current block. 
eth / token balance - simple and straightforward for me.
approved amount - some more info needed

would need to dive in your code and understand what you really looking for so it will be beneficial to maybe have a small chat and discuss more on this.

Learn more [on the Gitcoin Issue Details page](https://gitcoin.co/issue/airalab/robonomics_comm/19/999).

2. [jlegha](https://gitcoin.co/profile/jlegha) has applied to start work _(Funders only: [approve worker](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?mutate_worker_action=approve&worker=jlegha) | [reject worker](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?mutate_worker_action=reject&worker=jlegha))_ (2nd dibs). 
    
    I am thinking for some methods to apply if they work.

Learn more [on the Gitcoin Issue Details page](https://gitcoin.co/issue/airalab/robonomics_comm/19/999).

gitcoinbot, 2018-08-25:
@Rohithzr Hello from Gitcoin Core - are you still working on this issue? Please submit a WIP PR or comment back within the next 3 days or you will be removed from this ticket and it will be returned to an ‘Open’ status. Please let us know if you have questions!
* [x] warning (3 days)
* [ ] escalation to mods (6 days)

Funders only: Snooze warnings for [1 day](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?snooze=1) | [3 days](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?snooze=3) | [5 days](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?snooze=5) | [10 days](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?snooze=10) | [100 days](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?snooze=100)

Rohithzr, 2018-08-26:
just a bit of festive weekend in my country, Ill be back tomorrow and hopefully a PR in a couple of days

Rohithzr, 2018-08-28:
@khssnv 
I have very little information about this project at the moment so please let me know if the below solutions are not correct or why.

1. current block : 
```
def findBlockNumber():
	blockNumber = web3.eth.getBlock('latest')
	isSyncing = web3.eth.syncing
	if !isSyncing:
		blockNumber = web3.eth.syncing.currentBlock
	return blockNumber
```

Rohithzr, 2018-08-29:
@khssnv which tokens are we supporting? ill make a function to get the balance. I would need some more elaboration on the `approved amount` issue.

khssnv, 2018-08-29:
Hi, @Rohithzr!

1. Please consider how ERC20 node API made. It provides [ROS Service](http://wiki.ros.org/ROS/Tutorials/WritingServiceClient%28python%29). You may also make a [topic publisher](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29) for periodical blockchain info,
2. We support ERC20 tokens. Account Ether balance service getter also necessary,
3. As I know, there is allowance erc20 function to check the rest of approved, you may use it.

gitcoinbot, 2018-09-09:
Issue Status: 1. Open 2. **Started** 3. Submitted 4. Done 

<hr>

@Rohithzr due to inactivity, we have escalated [this issue](https://gitcoin.co/issue/airalab/robonomics_comm/19/999) to Gitcoin's moderation team. Let us know if you believe this has been done in error!

* [x] warning (3 days)
* [x] escalation to mods (6 days)

Funders only: Snooze warnings for [1 day](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?snooze=1) | [3 days](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?snooze=3) | [5 days](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?snooze=5) | [10 days](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?snooze=10) | [100 days](https://gitcoin.co/issue/airalab/robonomics_comm/19/999?snooze=100)

ryan-shea, 2018-10-25:
@Rohithzr are you still working on this? Ryan from Gitcoin checking in.

closed 2018-10-26 by strdn

ryan-shea, 2018-11-08:
Hey @strdn, should we close out the bounty on Gitcoin for this one?

akru, 2018-11-08:
Hey @ryan-shea, yes, close please, issue already implemented.

ryan-shea, 2018-12-10:
@akru, issue closed!