airalab/robonomics_comm #21: Liability finishes with empty result

khssnv, 2018-09-07:
Liability contract executions finishes with empty rosbag result files.

After recieving `objective` messages I publish data to some topics and then call `liability/finish` service. But `result` rosbag file is empty.

For example, liability contract [0xbeC4f052D062ef662AccB9A55fAbdbF4DFF0dA3F](https://etherscan.io/address/0xbeC4f052D062ef662AccB9A55fAbdbF4DFF0dA3F) finished with result [QmZ4vJFzEn8mEmVBupMniszaQy8uR1WSfyF9oDhnQfuq28](https://ipfs.io/ipfs/QmZ4vJFzEn8mEmVBupMniszaQy8uR1WSfyF9oDhnQfuq28). This file is empty. The same you can find in 0x223778907979259Db6f58c4d4A3f9cBd4b9e6B14 liability contract and many previous completed by the same promisor.

I tried to insert a wait for 5 seconds between `objective` messages recieving and my messages publication. It did not help, I've got the same empty Qm...uq28 result. I tried to insert 30 seconds wait to look an `executor` node subscribing to my topics. As you can see below, there is no subscriber to my topic while executing liability.

`
[root@aira-dronedirect-01:~]# rostopic info /agent/result/droneid
Type: std_msgs/String
Publishers:
* /agent (http://aira-dronedirect-01:43673/)
Subscribers: None
`

Restart did not help as well. I use robonomics_liability node from airapkgs robonomics_comm package with all arguments default.

Please find executor log from two of my attempts below. You may find related messages searching for `QmZ4vJFzEn8mEmVBupMniszaQy8uR1WSfyF9oDhnQfuq28` string.

<details>
  <summary>executor-1.log</summary>
[rospy.client][INFO] 2018-09-04 17:59:05,767: init_node, name[/liability/executor], pid[17148]
[xmlrpc][INFO] 2018-09-04 17:59:05,769: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2018-09-04 17:59:05,770: Started XML-RPC server [http://aira-dronedirect-01:37361/]
[rospy.init][INFO] 2018-09-04 17:59:05,772: ROS Slave URI: [http://aira-dronedirect-01:37361/]
[rospy.impl.masterslave][INFO] 2018-09-04 17:59:05,772: _ready: http://aira-dronedirect-01:37361/
[rospy.registration][INFO] 2018-09-04 17:59:05,775: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2018-09-04 17:59:05,775: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2018-09-04 17:59:05,873: registered with master
[rospy.rosout][INFO] 2018-09-04 17:59:05,874: initializing /rosout core topic
[rospy.rosout][INFO] 2018-09-04 17:59:05,880: connected to core topic /rosout
[rospy.simtime][INFO] 2018-09-04 17:59:05,883: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2018-09-04 17:59:05,972: topic[/liability/incoming] adding connection to [http://aira-dronedirect-01:45699/], count 0
[rospy.internal][INFO] 2018-09-04 17:59:05,993: topic[/liability/result] adding connection to [/liability/listener], count 0
[rospy.internal][INFO] 2018-09-04 17:59:06,173: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2018-09-04 17:59:15,896: topic[/liability/current] adding connection to [/agent], count 0
[rospy.internal][INFO] 2018-09-04 17:59:40,479: topic[/liability/current] removing connection to /agent
[rospy.internal][INFO] 2018-09-04 17:59:45,902: topic[/liability/current] adding connection to [/agent], count 0
[rosout][INFO] 2018-09-04 18:01:31,205: Append 0x804f307998e3C50265FF796d90bB5529919223D7 to liability queue.
[rosout][INFO] 2018-09-04 18:01:31,206: Start work on liability 0x804f307998e3C50265FF796d90bB5529919223D7
[rospy.internal][INFO] 2018-09-04 18:01:31,257: topic[/agent/objective/email] adding connection to [/agent], count 0
[rospy.internal][INFO] 2018-09-04 18:01:32,258: topic[/agent/objective/droneid] adding connection to [/agent], count 0
[rospy.internal][INFO] 2018-09-04 18:01:34,024: topic[/liability/result] adding connection to [/agent], count 1
[rosout][INFO] 2018-09-04 18:01:34,231: Liability 0x804f307998e3C50265FF796d90bB5529919223D7 finished with QmZ4vJFzEn8mEmVBupMniszaQy8uR1WSfyF9oDhnQfuq28
[rospy.internal][INFO] 2018-09-04 18:01:34,537: topic[/liability/result] removing connection to /agent
[rosout][INFO] 2018-09-04 18:15:27,068: Append 0x586C0d2BB75CA75424DB157a31c2C22886Babce3 to liability queue.
[rosout][INFO] 2018-09-04 18:15:27,069: Start work on liability 0x586C0d2BB75CA75424DB157a31c2C22886Babce3
[rospy.internal][INFO] 2018-09-04 18:15:29,477: topic[/liability/result] adding connection to [/agent], count 1
[rosout][INFO] 2018-09-04 18:15:30,093: Liability 0x586C0d2BB75CA75424DB157a31c2C22886Babce3 finished with QmZ4vJFzEn8mEmVBupMniszaQy8uR1WSfyF9oDhnQfuq28
[rospy.internal][INFO] 2018-09-04 18:15:30,457: topic[/liability/result] removing connection to /agent
[rosout][WARNING] 2018-09-04 18:49:09,050: Liability 0x4D15bDF7cdC91DB9EcAB542b116cc03fE49cE37a is not for me, SKIP.
[rosout][WARNING] 2018-09-04 19:48:37,498: Liability 0x94b7AF6a5F756158373C4149C6385C16c7902D0E is not for me, SKIP.
[rosout][WARNING] 2018-09-04 20:48:41,063: Liability 0x0F2600AFe86B174617C5Ebac33F8DCc489348b0e is not for me, SKIP.
[rosout][WARNING] 2018-09-04 22:48:42,976: Liability 0x396375E508Ad42cb7C04643dBC596C0b07C351de is not for me, SKIP.
[rosout][WARNING] 2018-09-04 23:48:51,474: Liability 0xbAE153a4573Eff2c3e3cF6c22E9B4EA5F4bA204a is not for me, SKIP.
[rosout][WARNING] 2018-09-04 23:48:56,543: Liability 0xbAE153a4573Eff2c3e3cF6c22E9B4EA5F4bA204a is not for me, SKIP.
[rosout][WARNING] 2018-09-04 23:48:56,633: Liability 0xbAE153a4573Eff2c3e3cF6c22E9B4EA5F4bA204a is not for me, SKIP.
[rosout][WARNING] 2018-09-05 00:49:20,097: Liability 0xDD7f6Fd2FcA1743d36430dB333e24b4960F1DcbD is not for me, SKIP.
[rosout][WARNING] 2018-09-05 01:49:43,589: Liability 0x64C1ed8a56c4F8875D71B8A2437669434C624Ee9 is not for me, SKIP.
[rosout][WARNING] 2018-09-05 03:49:10,630: Liability 0x824C456f53eD55Ca0a7f9b4a86B9879924dC171A is not for me, SKIP.
[rosout][WARNING] 2018-09-05 04:49:34,278: Liability 0x4d1656090dE9c4bf9c6cA47DfFdBf50A66Ba4B13 is not for me, SKIP.
[rosout][WARNING] 2018-09-05 06:50:26,180: Liability 0x02E4caFD29fB7F3e11561499E4336748547EfA3f is not for me, SKIP.
[rosout][WARNING] 2018-09-05 07:49:54,620: Liability 0x69f5e7079e468390E1eCaFF17E24D55B95b44948 is not for me, SKIP.
[rosout][WARNING] 2018-09-05 08:48:58,109: Liability 0xc8305A2ff1350aafF8601BC813215c693302Ad5e is not for me, SKIP.
[rosout][WARNING] 2018-09-05 09:48:56,596: Liability 0xA70e5ddECD14175aCfb63318440C413839517366 is not for me, SKIP.
[rosout][WARNING] 2018-09-05 10:49:55,114: Liability 0x70dc0EFfa63484701e96faDAE3a08C3db8B2D51F is not for me, SKIP.
[rosout][WARNING] 2018-09-05 11:49:13,549: Liability 0x2B5962Fc164aC1d9f36279ad66D395A9089EA7a4 is not for me, SKIP.
[rosout][INFO] 2018-09-05 12:19:55,365: Append 0xB6B6765191Bba09Ee181704eD0bFA29EF089EbA7 to liability queue.
[rosout][INFO] 2018-09-05 12:19:55,367: Start work on liability 0xB6B6765191Bba09Ee181704eD0bFA29EF089EbA7
[... 241 lines cut ...]
[rospy.init][INFO] 2018-09-07 12:14:09,210: ROS Slave URI: [http://aira-dronedirect-01:39903/]
[rospy.impl.masterslave][INFO] 2018-09-07 12:14:09,210: _ready: http://aira-dronedirect-01:39903/
[rospy.registration][INFO] 2018-09-07 12:14:09,211: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2018-09-07 12:14:09,211: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2018-09-07 12:14:09,310: registered with master
[rospy.rosout][INFO] 2018-09-07 12:14:09,311: initializing /rosout core topic
[rospy.rosout][INFO] 2018-09-07 12:14:09,315: connected to core topic /rosout
[rospy.simtime][INFO] 2018-09-07 12:14:09,320: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2018-09-07 12:14:09,409: topic[/liability/incoming] adding connection to [http://aira-dronedirect-01:42985/], count 0
[rospy.internal][INFO] 2018-09-07 12:14:09,510: topic[/liability/result] adding connection to [/liability/listener], count 0
[rospy.internal][INFO] 2018-09-07 12:14:09,560: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2018-09-07 12:15:08,741: topic[/liability/current] adding connection to [/agent], count 0
[rosout][INFO] 2018-09-07 12:16:04,713: Append 0x223778907979259Db6f58c4d4A3f9cBd4b9e6B14 to liability queue.
[rosout][INFO] 2018-09-07 12:16:04,714: Start work on liability 0x223778907979259Db6f58c4d4A3f9cBd4b9e6B14
[rospy.internal][INFO] 2018-09-07 12:16:04,787: topic[/agent/objective/email] adding connection to [/agent], count 0
[rospy.internal][INFO] 2018-09-07 12:16:05,788: topic[/agent/objective/droneid] adding connection to [/agent], count 0
[... cut ...]
[rosout][INFO] 2018-09-07 12:19:50,004: Append 0xbeC4f052D062ef662AccB9A55fAbdbF4DFF0dA3F to liability queue.
[rosout][INFO] 2018-09-07 12:19:50,006: Start work on liability 0xbeC4f052D062ef662AccB9A55fAbdbF4DFF0dA3F
[rospy.internal][INFO] 2018-09-07 12:19:57,108: topic[/liability/result] adding connection to [/agent], count 1
[rosout][INFO] 2018-09-07 12:19:58,042: Liability 0xbeC4f052D062ef662AccB9A55fAbdbF4DFF0dA3F finished with QmZ4vJFzEn8mEmVBupMniszaQy8uR1WSfyF9oDhnQfuq28
[rospy.internal][INFO] 2018-09-07 12:19:58,081: topic[/liability/result] removing connection to /agent
</details>

akru, 2018-09-09:
Do you have Debug log of liability execution?

khssnv, 2018-09-10:
Yes, please find it below.

<details>
  <summary>~/.ros/log/latest/liability-executor-5.log</summary>

```
[root@aira-dronedirect-01:~]# cat ~/.ros/log/latest/liability-executor-5.log 
[rospy.client][INFO] 2018-09-10 13:28:55,759: init_node, name[/liability/executor], pid[30665]
[xmlrpc][INFO] 2018-09-10 13:28:55,760: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2018-09-10 13:28:55,761: Started XML-RPC server [http://aira-dronedirect-01:41743/]
[rospy.impl.masterslave][INFO] 2018-09-10 13:28:55,764: _ready: http://aira-dronedirect-01:41743/
[rospy.init][INFO] 2018-09-10 13:28:55,764: ROS Slave URI: [http://aira-dronedirect-01:41743/]
[rospy.registration][INFO] 2018-09-10 13:28:55,766: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2018-09-10 13:28:55,766: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2018-09-10 13:28:55,865: registered with master
[rospy.rosout][INFO] 2018-09-10 13:28:55,865: initializing /rosout core topic
[rospy.rosout][INFO] 2018-09-10 13:28:55,870: connected to core topic /rosout
[rospy.simtime][INFO] 2018-09-10 13:28:55,874: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2018-09-10 13:28:56,011: topic[/liability/incoming] adding connection to [http://aira-dronedirect-01:41907/], count 0
[rospy.internal][INFO] 2018-09-10 13:28:56,103: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2018-09-10 13:28:56,113: topic[/liability/result] adding connection to [/liability/listener], count 0
[rospy.internal][INFO] 2018-09-10 13:31:48,132: topic[/liability/current] adding connection to [/agent], count 0
[rosout][INFO] 2018-09-10 13:58:52,950: Append 0x1092702873ac9c7B6f9Cb8fB575F26125F2AFd9a to liability queue.
[rosout][INFO] 2018-09-10 13:58:52,952: Start work on liability 0x1092702873ac9c7B6f9Cb8fB575F26125F2AFd9a
[rosout][DEBUG] 2018-09-10 13:58:52,955: Temporary directory created: /tmp/tmpkvwuv_6f
[rosout][DEBUG] 2018-09-10 13:58:52,955: Getting objective QmPfwdqsFt6orDEj1tUN8aqNdKrPU52NgsdaTaYMphJWyi...
[rosout][DEBUG] 2018-09-10 13:58:52,965: Objective is written to /tmp/tmpkvwuv_6f/QmPfwdqsFt6orDEj1tUN8aqNdKrPU52NgsdaTaYMphJWyi
[rosout][DEBUG] 2018-09-10 13:58:52,966: Start recording to /tmp/tmpkvwuv_6f/result.bag...
[rosout][DEBUG] 2018-09-10 13:58:52,968: Rosbag recorder started
[rosout][DEBUG] 2018-09-10 13:58:52,970: Player created for QmPfwdqsFt6orDEj1tUN8aqNdKrPU52NgsdaTaYMphJWyi
[rosout][DEBUG] 2018-09-10 13:58:52,973: Player started
[rosout][DEBUG] 2018-09-10 13:58:52,974: Rosbag player started
[rosout][DEBUG] 2018-09-10 13:58:52,982: New publisher /agent/objective/email of <class 'tmpwqszp78o._std_msgs__String'>
[rospy.internal][INFO] 2018-09-10 13:58:53,029: topic[/agent/objective/email] adding connection to [/agent], count 0
[rosout][DEBUG] 2018-09-10 13:58:53,987: Publish message data: "a.khssnv@gmail.com" to /agent/objective/email
[rosout][DEBUG] 2018-09-10 13:58:53,989: New publisher /agent/objective/droneid of <class 'tmpwqszp78o._std_msgs__String'>
[rospy.internal][INFO] 2018-09-10 13:58:54,030: topic[/agent/objective/droneid] adding connection to [/agent], count 0
[rosout][DEBUG] 2018-09-10 13:58:54,995: Publish message data: "AL76" to /agent/objective/droneid
[rosout][DEBUG] 2018-09-10 13:58:59,998: Publish done
[rospy.internal][INFO] 2018-09-10 13:59:00,754: topic[/liability/result] adding connection to [/agent], count 1
[rosout][INFO] 2018-09-10 13:59:00,991: Liability 0x1092702873ac9c7B6f9Cb8fB575F26125F2AFd9a finished with QmZ4vJFzEn8mEmVBupMniszaQy8uR1WSfyF9oDhnQfuq28
[rospy.internal][INFO] 2018-09-10 13:59:01,452: topic[/liability/result] removing connection to /agent
```

</details>

akru, 2018-09-11:
#23 possible fix it, can you check?