airalab/robonomics_comm #22: Nodes crashes after a fail to connect web3 HTTP provider

khssnv, 2018-09-07:
Running AIRA for a day or longer I often face with `robonomics_liability` and `ethereum_common` crashes at least once a day with an error like:
`
requests.exceptions.ConnectionError: HTTPConnectionPool(host='127.0.0.1', port=8545): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f986007fa20>: Failed to establish a new connection: [Errno 111] Connection refused',))
`
After it, for example, `erc20` node services becomes unavailable. I did not check `robonomics_liability`. And at the same time `parity` still works ok and if I restsart `robonomics_comm` packages it works all right.

I suggest to handle this error with user warning notification in logs and reconnect to RPC after it becomes available.

akru, 2018-09-09:
It looks like Ethereum node unavailable, do you have Ethereum node logs?

khssnv, 2018-09-11:
I just catch similar situation, please find a log below. In that case parity stops syncing, restart helped. But it seems like it's not what I told in opening comment where parity continue writing a log. I will collect more information.

```
[root@aira-mainnet-dev:~]# journalctl -u parity -f
-- Logs begin at Mon 2018-09-10 19:49:54 +04. --
Sep 11 00:52:18 aira-mainnet-dev 91c9mq6jkwmsh69q82fk6zb9vr1v0q54-unit-script-parity-start[26912]: 2018-09-11 00:52:18  Imported #6308139 0x42ae…0d19 (8.01 Mgas)                                          
Sep 11 00:52:20 aira-mainnet-dev 91c9mq6jkwmsh69q82fk6zb9vr1v0q54-unit-script-parity-start[26912]: 2018-09-11 00:52:20    26/50 peers     10 MiB cache  0 bytes queue  RPC:  0 conn,  0 req/s,   0 µs      
Sep 11 00:52:25 aira-mainnet-dev 91c9mq6jkwmsh69q82fk6zb9vr1v0q54-unit-script-parity-start[26912]: 2018-09-11 00:52:25  Syncing #6308139 0x42ae…0d19     0 hdr/s      0+    0 Qed  #6308139   26/50 peers  
  10 MiB cache  0 bytes queue  RPC:  0 conn,  0 req/s,   0 µs
Sep 11 00:52:25 aira-mainnet-dev 91c9mq6jkwmsh69q82fk6zb9vr1v0q54-unit-script-parity-start[26912]: 2018-09-11 00:52:25  Imported #6308140 0x2ec5…84a9 (8.00 Mgas)                                          
Sep 11 00:52:44 aira-mainnet-dev 91c9mq6jkwmsh69q82fk6zb9vr1v0q54-unit-script-parity-start[26912]: 2018-09-11 00:52:44  Imported #6308141 0xf989…c31d (7.99 Mgas)                                          
Sep 11 00:52:50 aira-mainnet-dev 91c9mq6jkwmsh69q82fk6zb9vr1v0q54-unit-script-parity-start[26912]: 2018-09-11 00:52:50  Imported #6308142 0x5866…bdbc (7.90 Mgas)                                          
Sep 11 00:52:55 aira-mainnet-dev 91c9mq6jkwmsh69q82fk6zb9vr1v0q54-unit-script-parity-start[26912]: 2018-09-11 00:52:55    26/50 peers     10 MiB cache  0 bytes queue  RPC:  0 conn,  0 req/s,   0 µs      
Sep 11 00:53:13 aira-mainnet-dev 91c9mq6jkwmsh69q82fk6zb9vr1v0q54-unit-script-parity-start[26912]: 2018-09-11 00:53:13  Imported #6308143 0x3cfe…128f (7.92 Mgas)                                          
Sep 11 00:53:25 aira-mainnet-dev 91c9mq6jkwmsh69q82fk6zb9vr1v0q54-unit-script-parity-start[26912]: 2018-09-11 00:53:25  Imported #6308144 0x8a2f…0b3e (7.15 Mgas)                                          
Sep 11 00:53:25 aira-mainnet-dev 91c9mq6jkwmsh69q82fk6zb9vr1v0q54-unit-script-parity-start[26912]: 2018-09-11 00:53:25    26/50 peers     10 MiB cache  0 bytes queue  RPC:  0 conn,  0 req/s,   0 µs 
```

akru, 2018-09-26:
@strdn looks like a temporary connection problems, Ethereum interaction should be wrapped in ‘try’blocks.

commit db3ff2e9 (2018-10-02): do not fail lighthouse/liability if parity is temporary unavailable

fix #29, fix #22
closed 2018-10-02 by akru