airalab/robonomics_comm #24: Add an ability to process several liability contracts simultaneously by executor

valery-l, 2018-09-13:
As we have found, there is a limitation at executor, that results in ability to process only one liability simultaneously. They are taken one by one (executor.py:62), and there is no ability to distinguish what exact liability is finished (executor.py:54).
We are making an integration with our U-traffic drone base (https://sim-labs.com/industries/u-traffic) and this base could receive new ask while some others are not finished yet.

akru, 2018-09-14:
You are absolutely right, currently executor dosn’t support parallelism for liabilities. It’s possible but not implemented yet.

I keep in mind your improvement proposal. On the first view executor may start liabilities in isolated ros namespace in future releases.

khssnv, 2018-09-30:
@akru, any open conceptual questions here or implementation could be made in a next release?

closed 2018-10-26 by strdn