airalab/robonomics_comm #26: Parity light client

tuuzdu, 2018-09-16:
I have issue with `robonomics_liability` and Parity light client.
Is it possible to use light clients?
```
Exception in thread Thread-12:
Traceback (most recent call last):
  File "/nix/store/hpsfqcw18x66bq6175m1iq2wp7w90vi4-python3-3.6.6/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/nix/store/hpsfqcw18x66bq6175m1iq2wp7w90vi4-python3-3.6.6/lib/python3.6/threading.py", line 1182, in run
    self.function(*self.args, **self.kwargs)
  File "/home/de/ws_catkin/src/robonomics_comm/robonomics_liability/src/robonomics_liability/listener.py", line 80, in liability_filter_thread
    for entry in liability_filter.get_new_entries():
  File "/nix/store/kdjyaidk6y5s33x9fk0q3k9pzhzp0nig-python3.6-web3-4.5.0/lib/python3.6/site-packages/web3/utils/filters.py", line 105, in get_new_entries
    log_entries = self._filter_valid_entries(self.web3.eth.getFilterChanges(self.filter_id))
  File "/nix/store/kdjyaidk6y5s33x9fk0q3k9pzhzp0nig-python3.6-web3-4.5.0/lib/python3.6/site-packages/web3/eth.py", line 344, in getFilterChanges
    "eth_getFilterChanges", [filter_id],
  File "/nix/store/kdjyaidk6y5s33x9fk0q3k9pzhzp0nig-python3.6-web3-4.5.0/lib/python3.6/site-packages/web3/manager.py", line 110, in request_blocking
    raise ValueError(response["error"])
ValueError: {'code': -32000, 'message': 'One of the blocks specified in filter (fromBlock, toBlock or blockHash) cannot be found', 'data': 'latest'}
```

akru, 2018-09-16:
It looks as parity/Web3 specified error.

Currently light client isn’t officially supported by us because it should more tests with robonomics_comm.

khssnv, 2018-09-16:
One may find my experience with light client useful. Assuming at the moment, it's not useful.
https://github.com/airalab/aira/issues/46

Maybe we can implement public node connection option to answer a need of lighter ethereum interface.

akru, 2018-10-03:
Could you try to use `robonomics_liability` without parity using https://github.com/airalab/robonomics_comm/pull/35

khssnv, 2018-10-06:
It works with my own public node while seems like `infura` node don't provide a RPC service necessary for listener, you may find a log below. Signer can sign and send something to IPFS. Few issues I faced with while use it with my public node I'll describe outside this thread.

<details>
  <summary>liability-listener-4.log</summary>

[rospy.client][INFO] 2018-10-05 23:03:40,965: init_node, name[/liability/listener], pid[26781]
[xmlrpc][INFO] 2018-10-05 23:03:40,979: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2018-10-05 23:03:40,980: Started XML-RPC server [http://nixos:38479/]
[rospy.init][INFO] 2018-10-05 23:03:40,987: ROS Slave URI: [http://nixos:38479/]
[rospy.impl.masterslave][INFO] 2018-10-05 23:03:40,989: _ready: http://nixos:38479/
[xmlrpc][INFO] 2018-10-05 23:03:41,003: xml rpc node: starting XML-RPC server
[rospy.registration][INFO] 2018-10-05 23:03:41,004: Registering with master node http://localhost:11311
[rospy.init][INFO] 2018-10-05 23:03:41,090: registered with master
[rospy.rosout][INFO] 2018-10-05 23:03:41,091: initializing /rosout core topic
[rospy.rosout][INFO] 2018-10-05 23:03:41,101: connected to core topic /rosout
[rospy.simtime][INFO] 2018-10-05 23:03:41,115: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2018-10-05 23:03:41,362: topic[/rosout] adding connection to [/rosout], count 0
[rosout][WARNING] 2018-10-05 23:03:46,740: Failed to create liability filter with exception: "405 Client Error: Method Not Allowed for url: https://mainnet.infura.io/9omcLOJTvmNvyQ45rybj"
[rospy.internal][INFO] 2018-10-05 23:03:46,917: topic[/liability/infochan/signing/result] adding connection to [/liability/infochan/signer], count 0
[rosout][ERROR] 2018-10-05 23:03:46,926: listener liability filter exception: 'Listener' object has no attribute 'liability_filter'
[rosout][WARNING] 2018-10-05 23:03:48,946: Failed to create liability filter with exception: "405 Client Error: Method Not Allowed for url: https://mainnet.infura.io/9omcLOJTvmNvyQ45rybj"
[rospy.core][INFO] 2018-10-05 23:03:49,607: signal_shutdown [signal-2]
[rospy.internal][INFO] 2018-10-05 23:03:49,661: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2018-10-05 23:03:49,667: topic[/liability/infochan/signing/result] removing connection to /liability/infochan/signer
[rospy.impl.masterslave][INFO] 2018-10-05 23:03:49,704: signal-2
[rosout][ERROR] 2018-10-05 23:03:53,949: listener liability filter exception: 'Listener' object has no attribute 'liability_filter'
[rosout][WARNING] 2018-10-05 23:03:55,804: Failed to create liability filter with exception: "405 Client Error: Method Not Allowed for url: https://mainnet.infura.io/9omcLOJTvmNvyQ45rybj"
[rosout][ERROR] 2018-10-05 23:04:00,827: listener liability filter exception: 'Listener' object has no attribute 'liability_filter'
[rosout][WARNING] 2018-10-05 23:04:02,868: Failed to create liability filter with exception: "405 Client Error: Method Not Allowed for url: https://mainnet.infura.io/9omcLOJTvmNvyQ45rybj"
[rospy.core][INFO] 2018-10-05 23:04:04,698: signal_shutdown [signal-15]
[rospy.core][INFO] 2018-10-05 23:04:04,735: signal_shutdown [atexit]
</details>

<details>
  <summary>liability-infochan-signer-3.log</summary>

[rospy.client][INFO] 2018-10-05 23:03:40,095: init_node, name[/liability/infochan/signer], pid[26780]
[xmlrpc][INFO] 2018-10-05 23:03:40,107: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2018-10-05 23:03:40,107: Started XML-RPC server [http://nixos:45167/]
[rospy.impl.masterslave][INFO] 2018-10-05 23:03:40,108: _ready: http://nixos:45167/
[rospy.init][INFO] 2018-10-05 23:03:40,119: ROS Slave URI: [http://nixos:45167/]
[rospy.registration][INFO] 2018-10-05 23:03:40,121: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2018-10-05 23:03:40,159: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2018-10-05 23:03:40,223: registered with master
[rospy.rosout][INFO] 2018-10-05 23:03:40,223: initializing /rosout core topic
[rospy.rosout][INFO] 2018-10-05 23:03:40,271: connected to core topic /rosout
[rospy.simtime][INFO] 2018-10-05 23:03:40,335: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2018-10-05 23:03:40,696: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2018-10-05 23:03:40,852: topic[/liability/infochan/sending/ask] adding connection to [/liability/infochan/channel], count 0
[rospy.internal][INFO] 2018-10-05 23:03:40,914: topic[/liability/infochan/sending/bid] adding connection to [/liability/infochan/channel], count 0
[rospy.internal][INFO] 2018-10-05 23:03:40,940: topic[/liability/infochan/sending/result] adding connection to [/liability/infochan/channel], count 0
[rospy.internal][INFO] 2018-10-05 23:03:46,899: topic[/liability/infochan/signing/result] adding connection to [http://nixos:38479/], count 0
[rospy.internal][INFO] 2018-10-05 23:03:48,694: topic[/liability/infochan/signing/ask] adding connection to [http://nixos:33297/], count 0
[rospy.internal][INFO] 2018-10-05 23:03:48,776: topic[/liability/infochan/signing/bid] adding connection to [http://nixos:33297/], count 0
[rosout][INFO] 2018-10-05 23:03:49,179: askhash: b'026c52d28d9fb62a1ef965988def812f888e2fe922af7124494760c89c1f665d' signature: b'd68a2b2f6730f6da38a7674a8445e22eb0b3f4eb4b76c8c9d97913016e22ff6e29f09eade$
[... cut ...]
[rospy.internal][INFO] 2018-10-05 23:03:49,752: topic[/liability/infochan/sending/result] removing connection to /liability/infochan/channel
[rospy.internal][INFO] 2018-10-05 23:03:49,772: topic[/liability/infochan/signing/ask] removing connection to http://nixos:33297/
[rospy.internal][INFO] 2018-10-05 23:03:49,776: topic[/liability/infochan/signing/bid] removing connection to http://nixos:33297/
[rospy.impl.masterslave][INFO] 2018-10-05 23:03:49,777: signal-2
[rospy.core][INFO] 2018-10-05 23:03:50,289: signal_shutdown [atexit]
</details>

akru, 2018-10-22:
Hi @tuuzdu, is this issue actual for you?

tuuzdu, 2018-10-23:
In near future I don't need support of light client. You can close this issue.

strdn, 2018-10-23:
@tuuzdu, with coming release (https://github.com/airalab/robonomics_comm/releases/tag/v0.3) you can use remote parity node such as infura.io or your own.

closed 2018-10-23 by strdn

tuuzdu, 2018-10-28:
@strdn, what about support of "filter" by infura.io Ethereum API?
https://ethereum.stackexchange.com/questions/41653/cannot-create-filter-on-infura-via-web3-py
Is it not potential problem?

akru, 2018-10-28:
@tuuzdu it really work now, we are run test full user scenario using infura endpoint.

tuuzdu, 2018-10-28:
@tuuzdu okey, I will try infura.