airalab/robonomics_comm #33: Listener node fails with web3 BadFunctionCallOutput

khssnv, 2018-10-01:
While parity synced `robonomics_liability` service fails, please consider a log below. I'll try to collect more data to specity reproduction steps.

[image]

akru, 2018-10-02:
Are your chain is fully synchronized?

khssnv, 2018-10-02:
When I find it, parity had latest block number. I'd like to have a tool for logs capture to take it all automatically even when I have no time to leave system unavailable. Are they all in /var/lib/<service_user>/logs or you like to have any more on failure occurred?

akru, 2018-10-22:
Currently this error should not crash service, you are free to reopen issue when more data will collected.

closed 2018-10-22 by akru