airalab/robonomics_comm #62: Finish and/or restart for interrupted liability execution

khssnv, 2018-10-31:
Now I use only short exection models no longer a minutes, but I hope to develop CPS with liabilities take days or weeks to complete. The main reason to make it short now because if something goes wrong while liability exection like application software error, server restart or batteries dischage - liability execution interrupts with no way to continue or even restart it.

I see there are open questions on how to **continue** interrupted execution with no missing older _report_ data, for example.

But maybe it's possible to implement **restart** execution or even just allow to finish own liability which is not finished yet?

khssnv, 2018-12-08:
Any updates for this topic?

akru, 2018-12-26:
Available on `robonomics_liability-nightly`.

strdn, 2018-12-26:
@akru, working on this issue in not finished yet.

akru, 2018-12-26:
Rly? Please let me known when finish.

strdn, 2018-12-26:
@akru, please see PR #89

akru, 2018-12-26:
@strdn released as `nightly`

strdn, 2018-12-26:
@khssnv, please see #89 description and feel free to ask any question about new behaviour

closed 2018-12-26 by strdn