airalab/robonomics_comm #67: Liability finish failure

akru, 2018-11-07:
```
[INFO] [1541598206.793270]: askhash: b'5d3a97ce1253b746c3c4429d4b4d900cee0a945c03bb97d80a14e72e0739ea7a' signature: b'a6d7efec13[15/759]a51a560b407e36fb333e6ed0642a3cc5cb5a5ac2eb9bf7315c157dc22df166e9fa529c43218b0b35b82d3097e5f743c21a323661907f0a1b'                       
[ERROR] [1541600217.993505211]: poll failed with error Interrupted system call
[INFO] [1541600564.144855]: bidhash: b'7ff5a0c9330f6d32c31d8e0eebbd8d7d83b5e20abab77d07f8f8e902fbc4d1e7' signature: b'3481e6e7f4185cfca3
a7b56c9960325be9981888e6eb54637797896c2640e6ff2796643fcaee95079db4d8193b07a4d8d31dc24947572abd9b1d629a2e5fe9571b'
[INFO] [1541600568.254078]: askhash: b'f75b423e47584ed7133de06273c24de24102bdd314e62108d0c7f9ea5771de4e' signature: b'153075e4b4f6e4f727
72a8c1c9db2147f33b2e80ff92cfae309b44f9360c8a2861d34e200568eac55f60be2f8344ebdf82737d5cba86aa456eccf5e261695d6c1c'
[INFO] [1541600578.143143]: Append 0x114C41D69023328F9817F73082546eEF25ea7Be3 to liability queue.
[INFO] [1541600578.143657]: Prepare to start liability 0x114C41D69023328F9817F73082546eEF25ea7Be3
[INFO] [1541600627.034810]: Liability 0x114C41D69023328F9817F73082546eEF25ea7Be3 started
[ERROR] [1541600650.135887]: Error processing request: 'NoneType' object has no attribute 'stop'
```

**Reproduce**

1. clone github master
2. publish `test_ask`
3. publish `test_bic`
4. catch liability
5. start liability via service
6. stop liability via service

strdn, 2018-11-08:
@akru, more logs and info is required, I can't reproduce this issue:
```
[INFO] [1541660923.193153]: Liability 0x6624C3B54F5CF858A3c200199ceb7F5b1831DB56 started
[WARN] [1541660923.216513]: Objective is written to /tmp/tmpy9wlk_sp/Qmb3H3tHZ1QutcrLq7WEtQWbEWjA11aPqVmeatMSrmFXvE
[INFO] [1541660950.213960]: Liability 0x6624C3B54F5CF858A3c200199ceb7F5b1831DB56 finished with multihash: "QmYBLKsvBT4CkwRYidHjGrWkZMLaiXEcgJDTHe5MYgp61X"
[INFO] [1541660950.260743]: reshash: b'01d62c884c80c34d884840e0e9829d23a3497f0e45295b5b7c276bbbb345b8c7' signature: b'c5c6886f303a112a0f53398ec45f8172b5e0c12e320c53679cf26831943d42732d69f9ae3647f227d08e4d7191503b9f66dfeec720f189158f62adf08cdf219e1c'
```

akru, 2018-11-08:
As I see the error about Recorder instance, could you check it an I’ll try to cellect more data. Currently it’s not work in my laptop at all.

closed 2018-11-10 by akru