airalab/robonomics_comm #72: The messages isn't published to /liability/ready and /liability/incoming

AlexeiOvcharov, 2018-11-21:
I use robonomics beta 3 communication stack. After contract is successfully created I didn't see any messages in topics `/liability/ready` and `/liability/incoming`.

First, I created offer message:
```
model:
  multihash: "QmWZsRae58t2jnVcHaUBYrWMdaM6vwvpSs264Cmg3MrwY6"
objective:
  multihash: "QmY9dUwCrytB4unmU4JmA8DnjVj4sBxuNVXhPLJXowaTgx"
token: "0x98DFc5247da729c045D4938bABD081A61a455776"
cost: 1
validator: "0x0000000000000000000000000000000000000000"
lighthouse: "0xD06Eda15582ce766000f0e88Ef94E34217bf8fB7"
lighthouseFee: 0
deadline: 6745415
nonce: b''
signature: b''
```
Next, DApp sent next message:
```
model:
  multihash: "QmWZsRae58t2jnVcHaUBYrWMdaM6vwvpSs264Cmg3MrwY6"
objective:
  multihash: "QmY9dUwCrytB4unmU4JmA8DnjVj4sBxuNVXhPLJXowaTgx"
token: "0x98DFc5247da729c045D4938bABD081A61a455776"
cost: 1
lighthouse: "0xD06Eda15582ce766000f0e88Ef94E34217bf8fB7"
validator: "0x0000000000000000000000000000000000000000"
validatorFee: 0
deadline: 6745415
nonce: b'z\xb6\xc1\x08u\x04!\x9f\x8c\x86\x9ey\xf1Np\xc4=Tn#\x80\xb9+G\xb1\x07\x16\xa5\xaa\xbe\xe8\xce'
signature: b'\xeb\x8d\xefJ!$\x13\xfb\x1c\xd8_\xee\xdf\xc1\xf4\x88o\xf4\x82\x8c2\xa7\xad\x06\xd7\x9fhV\xd3\x8c~\x14 \r\xcan\x08[\x10N\xc0.\xca\xab\xee\xa4y\xf4\xa8\x8f\xa5\xfb\x05\xb0\xd7B\xdb\xd7\x0cd\xaf_\x87/\x1c'
```
And contract was successfully crated.
(https://etherscan.io/tx/0xd27e8890195c71668e82f187ac89c0b027915acc81d0a37d50fa8592a470296b)

**configuration.nix**
```nix
{ config, pkgs, ... }:

{
    # Include the results of the hardware scan.
    imports = [ ./hardware-configuration.nix ];

    # Use the GRUB 2 boot loader.
    boot.loader.grub.enable = true;
    boot.loader.grub.version = 2;

    boot.initrd.availableKernelModules = [ "virtio_net" "virtio_pci" "virtio_blk" "9p" "9pnet_virtio" "virtio-scsi" ];
    boot.initrd.kernelModules = [ "virtio_balloon" "virtio_console" "virtio_rng" "virtio_scsi" "sym53c8xx" "scsi_mod" "sd_mod" "sg" ];
    boot.kernelModules = [ "kvm_intel" ];

    boot.loader.grub.device = "/dev/sda"; # or "nodev" for efi only

    networking.hostName = "kuka-agent"; # Define your hostname.
    # networking.wireless.enable = true;  # Enables wireless support via wpa_supplicant.
    networking.extraHosts = ''
        192.168.12.4 kuka-agent
        192.168.12.3 kuka-01
    '';

    # Select internationalisation properties.
    i18n = {
        consoleFont = "Lat2-Terminus16";
        consoleKeyMap = "us";
        defaultLocale = "en_US.UTF-8";
    };

    # Set your time zone.
    time.timeZone = "Europe/Samara";

[... 94 lines cut ...]
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 984: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 985: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 986: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 987: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 988: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 989: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 990: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 991: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 992: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 993: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 994: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 995: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 996: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 997: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 998: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 999: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 1000: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 1001: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 1002: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 1003: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 1004: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/gcqjnwn6mimg8gqfq7s67avjhsibz9mk-rosbash-1.14.1/share/rosbash/rosbash: line 1005: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/7870vlzkdmqhgf7cll4jmlrbp0644f6b-catkin-0.7.11/etc/catkin/profile.d/05.catkin_make.bash: line 67: complete: command not found
Nov 21 14:42:02 kuka-agent jiyz22h8k0qfli85g2pir0hk9liswrkp-unit-script-liability-start[1642]: /nix/store/7870vlzkdmqhgf7cll4jmlrbp0644f6b-catkin-0.7.11/etc/catkin/profile.d/05.catkin_make_isolated.bash: line 65: complete: command not found
```

akru, 2018-11-21:
No error in this issue, unfortunately I can't reproduce your error.

strdn, 2018-11-21:
@AlexeiOvcharov, first add mainnet ens address:
liability.ens = "0x314159265dD8dbb310642f98f50C066173C1259b"

akru, 2018-11-23:
Fixed at https://github.com/airalab/airapkgs/commit/4dee8aef9be60c42816fb582f3a149b6e6a8e6ef

AlexeiOvcharov, 2018-11-23:
I did `nixos-rebuild switch -I nixpkgs=airapkgs/` and tried again. Errors were found in log file `/var/lib/liability/.ros/log/0312c8a2-ef73-11e8-bb13-000c298f680e/liability-listener-3.log`:

```
[rospy.client][INFO] 2018-11-24 02:56:53,719: init_node, name[/liability/listener], pid[2930]
[xmlrpc][INFO] 2018-11-24 02:56:53,720: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2018-11-24 02:56:53,721: Started XML-RPC server [http://kuka-agent:32991/]
[rospy.init][INFO] 2018-11-24 02:56:53,722: ROS Slave URI: [http://kuka-agent:32991/]
[rospy.impl.masterslave][INFO] 2018-11-24 02:56:53,722: _ready: http://kuka-agent:32991/
[rospy.registration][INFO] 2018-11-24 02:56:53,724: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2018-11-24 02:56:53,725: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2018-11-24 02:56:53,823: registered with master
[rospy.rosout][INFO] 2018-11-24 02:56:53,823: initializing /rosout core topic
[rospy.rosout][INFO] 2018-11-24 02:56:53,828: connected to core topic /rosout
[rospy.simtime][INFO] 2018-11-24 02:56:53,832: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2018-11-24 02:56:53,957: topic[/liability/incoming] adding connection to [/liability/executor], count 0
[rospy.internal][INFO] 2018-11-24 02:56:54,040: topic[/rosout] adding connection to [/rosout], count 0
[rosout][WARNING] 2018-11-24 02:57:17,285: Failed to create liability filter with exception: ""
[rosout][ERROR] 2018-11-24 02:57:17,337: listener liability filter exception: 'Listener' object has no attribute 'liability_filter'
[rospy.internal][INFO] 2018-11-24 02:57:17,349: topic[/liability/result] adding connection to [http://kuka-agent:37539/], count 0
[rospy.internal][INFO] 2018-11-24 02:57:17,359: topic[/liability/infochan/eth/signing/result] adding connection to [/liability/infochan/eth/signer], count 0
[rospy.internal][INFO] 2018-11-24 02:59:20,422: topic[/liability/incoming] adding connection to [/rostopic_12240_1543013960204], count 1
[rosout][ERROR] 2018-11-24 03:04:38,353: listener liability filter exception: Could not transact with/call contract function, is contract deployed correctly and chain synced?
[rosout][ERROR] 2018-11-24 03:13:36,550: listener liability filter exception: Could not decode contract function call promisee return data b'' for output_types ['address']
[rosout][ERROR] 2018-11-24 03:15:26,388: listener liability filter exception: Could not decode contract function call model return data b'' for output_types ['bytes']
[rosout][ERROR] 2018-11-24 03:16:46,618: listener liability filter exception: Could not decode contract function call validatorFee return data b'' for output_types ['uint256']
```

akru, 2018-11-24:
It looks like blockchain read failure, maybe node connection is not so good. Could you try to use local Ethereum node?

closed 2018-12-01 by akru