airalab/robonomics_comm #74: multiple files named [liability.launch] in package [robonomics_liability]

tuuzdu, 2018-11-25:
After 
```
nix build -f release.nix
. ws/devel/setup.bash
roslaunch robonomics_liability liability.launch
```
on NixOS or if used Nix package manager, appear trouble below:
```
[de@nixos:~/robonomics_comm]$ roslaunch robonomics_liability liability.launch 
RLException: multiple files named [liability.launch] in package [robonomics_liability]:
- /home/de/robonomics_comm/result/share/robonomics_liability/launch/liability.launch
- /nix/store/sh8ljpq5wy1vqmp9z4adb9ck7xfi6b42-robonomics_comm-master/share/robonomics_liability/launch/liability.launch
Please specify full path instead
The traceback for the exception was written to the log file
```

akru, 2018-11-26:
Currently you can run launch file directly, I'm working on similar issues around ros package finder.

akru, 2018-12-26:
Fixed

closed 2018-12-26 by akru