airalab/robonomics_comm #77: No packages in environment after a build in catkin workspace

khssnv, 2018-12-01:
**Expected behavior**
Packages from `robonomics_comm` available to launch after a build by `cakin_make` and sourcing from `setup.bash` file.

**Observed behafior**
No `robonomics_comm` packages in environment after a build in catkin workspace.

**Package `robonomics_comm` version**
Tag [v0.4](https://github.com/airalab/robonomics_comm/tree/v0.4)

**How to reproduce**
Please see a log below.

<details>
  <summary>Building & launching</summary>

```bash
[00:52][aira-05][root][~]# ls ~/.nix-profile
manifest.nix
[000:52][aira-05][root][~]# nix-env -i robonomics_dev
installing 'robonomics_dev-0.0.0'
building '/nix/store/zsnaf2dc829i4523prhna78p7bvkzpw9-user-environment.drv'...
created 7 symlinks in user environment
[00:52][aira-05][root][~]# ls ~/catkin_ws/
src
[00:52][aira-05][root][~]# ls ~/catkin_ws/src
CMakeLists.txt  robonomics_comm
[00:52][aira-05][root][~]# nix-shell -p gcc
[nix-shell:~]# cd catkin_ws/
[nix-shell:~/catkin_ws]# . ~/.nix-profile/setup.bash
[nix-shell:~/catkin_ws]# catkin_make
Base path: /root/catkin_ws
Source space: /root/catkin_ws/src
Build space: /root/catkin_ws/build
Devel space: /root/catkin_ws/devel
Install space: /root/catkin_ws/install
####
#### Running command: "cmake /root/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/root/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/root/catkin_ws/install -G Unix Makefiles" in "/root/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /root/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /root/.nix-profile;/nix/store/2x6iv4zw7d3if9i549kaw0y91g8d39iy-patchelf-0.9;/nix/store/wvz8gp7bhwp572jy2w1n48n1rn8wdrw3-paxctl-0.9;/nix/store/10yq7kwlvbc6h658izmrlsspry1g9f3c-gcc-wrapper-7.3.0;/nix/store/an3icm1hajqx1lld6n8q4qr40nwbs544-binutils-wrapper-2.30;/nix/store/bwfygfcdvis9wd1c1v51xwnwhw1hx0a0-python3-3.6.6;/nix/store/sxxy01wn82n7wjl5a384344m2i4xszx6-cmake-3.12.1;/nix/store/1h6ihpgpm3866rni27737fcfwlbalrbh-python3.6-catkin_pkg-0.4.9;/nix/store/0j3p5qrhcazg8lks7fdniz9snznfrlgy-python3.6-pyparsing-2.2.2;/nix/store/k99nwwi4nmwshaj40yy845jpr1gzm35r-python3.6-setuptools-40.2.0;/nix/store/1vdlxz3d0i6r06hjagig900w5brwg327-python3.6-docutils-0.14;/nix/store/gm0qfvbivrsqfvbfdb7igvp0qdlgl9r5-python3.6-python-dateutil-2.7.3;/nix/store/gh75sa1lkavznxqv9qbsbvqbkx4j79zs-python3.6-six-1.11.0;/nix/store/92bpv9zz1v5pvp67jzgnqx19jr545jiz-python3.6-setuptools_scm-3.1.0;/nix/store/w7nx3caffpbbp1xyawz77ry1brb9440d-python3.6-rospkg-1.1.7;/nix/store/c7mw8rk4wr52gghfzcmakan14x9zw32j-python3.6-PyYAML-3.13;/nix/store/yx2zlz515qd3xw8fccajg9wgmw2bc0yr-libyaml-0.2.1;/nix/store/w0xwwcdhjglx5bznp0wqmximins7cp4p-catkin-0.7.14;/nix/store/kg52fzv2qm94yp5rdriq9cbc8hdg33yr-empy-3.3.2;/nix/store/z18pqfw051x9zhg86zxjiilq4sy8vk1a-ros_comm-1.14.4;/nix/store/7n19gwyhh5sr3iqa0pmz16x8d9znf86j-message_filters-1.13.6;/nix/store/fxir3akhaisyihpcpdzz4cvhbgrgzxln-rosconsole-1.14.4;/nix/store/s42jn6v0rslfixk6jpdldc5f8xl3mlr1-boost-1.67_0-dev;/nix/store/z9iy0r4w0dhgq19yrxxbky76p1cfgr1r-boost-1.67_0;/nix/store/yn77gxnmcx5q4ypdi4yiszx172shn5wq-cpp_common-0.6.11;/nix/store/py5i8cxnlvb6qvcjvcq5vnn65lpp3rjy-console_bridge-0.3.2;/nix/store/kphlj8v2ja6zqpq770i1s4g7bcxbxpr6-log4cxx-0.10.0;/nix/store/wm7mrgsavbb7lridkq01nwbz19zikiz9-rosbuild-1.14.4;/nix/store/q35xxf9savinkkv1yy48cyqw6m6sikrc-message_generation-0.4.0;/nix/store/mda8g5hihw43kwk1pgci2fzl2c6lmbqq-genmsg-0.5.11;/nix/store/bw7cy1w0v54vh5i5ci4bngsybh7d30wh-gencpp-0.6.0;/nix/store/dj9mbf74jx0lrbmjg6ipykfrwzvcywxn-genpy-0.6.7;/nix/store/s8sjslzsf7q5m1fla8m5dksx5jyz2s9g-message_runtime-0.4.12;/nix/store/ipnb3py2myxlbdsaxq3gngjn8lqx3biq-roscpp_serialization-0.6.11;/nix/store/xq70szm3bivkbvxg77vh3mhnjhlmlyc1-rostime-0.6.11;/nix/store/8m0n78nxal41vzf04jcpkky2kxh2ywh3-roscpp_traits-0.6.11;/nix/store/lwq07p1ikvnymj78zmyks64fbm8ip9ps-rosunit-1.14.4;/nix/store/d79q5pngd3wi5hd6zcfrk3jpxrlrarci-roslib-1.14.4;/nix/store/jwac8i0w7ash92nyw4hk9i5wlhrirbs1-rospack-2.5.2;/nix/store/i04mvq8q5bkkb55smavppd3xwj0p8glx-cmake_modules-0.4.1;/nix/store/iwr0pcsgwaks20nv84aml85fls4fjlm9-rostest-1.14.4;/nix/store/c8392q7dp1afrfc5hljnf366jakq0s1i-rosgraph-1.14.4;/nix/store/vvy2j3vy364yw6dynpjnsxlxnyhcarrk-python3.6-netifaces-0.10.6;/nix/store/cz62mfmv9spv1pcqjvn0gzgk0396pvmb-roslaunch-1.14.4;/nix/store/zb3s0h9j6i3h6kv2qmfyny6l2j2l9cq2-rosclean-1.14.4;/nix/store/sbgrqq38q0yksr4r3zd6kn61arjb00bz-rosgraph_msgs-1.11.2;/nix/store/v4cvrh194rqbc9zxqax93wsi85q558dd-std_msgs-0.5.11;/nix/store/6ks5aw65jqai22sxci7wbd71wmw47gdx-rosmaster-1.14.4;/nix/store/jjrlw0bjxg36kvv5y7blgxq6pgx2h987-rosout-1.14.4;/nix/store/96rmy4zl1ld4rh3sdx187gjcbvmqjnaz-roscpp-1.14.4;/nix/store/gh676k9dk5hiaknmcbx4d4a1jwij7lih-roslang-1.14.4;/nix/store/109qyi8yiadj7gvhjbql72kx6iljs8p3-xmlrpcpp-1.13.6;/nix/store/zmmg4qbd0cqw8q93akhrrgr8yfw0qx1g-rosparam-1.14.4;/nix/store/5lpjw4mf3sagk8ackdy315xdm9jdzsls-python3.6-defusedxml-0.5.0;/nix/store/hpz7qzdxa0l29w8s24lx74xsbr6rd2y4-rospy-1.14.4;/nix/store/68wxm285kd39rdw0gakjyjfjnn5jm81f-ros-1.14.4;/nix/store/z48q6rlg20pi3vf9j0adpp0cfr99jgbl-rosbash-1.14.4;/nix/store/9c76qr138lma4c8m610vsrkacaqxx4z8-rosboost_cfg-1.14.4;/nix/store/3gjxqsvny6zkbijs8ns9rqy3hqwbj9ip-roscreate-1.14.4;/nix/store/0pwjaz1kjfva7pi9skj06p5pdq4zj984-rosmake-1.14.4;/nix/store/2030lva1cbjqsjszs643qhc90zc00l1w-rosbag-1.14.4;/nix/store/gpk4aswnsv2hmyvl6vi0hslj5bi12wbi-rosbag_storage-1.14.4;/nix/store/4vhi250xlzizzw28ih2jp2i3c0snss0h-roslz4-1.14.4;/nix/store/8v7r9125wpdsj61dah037q534jlxz6jc-lz4-1.8.3-dev;/nix/store/cwr62s01dwdd6fa4cyh9k0frdz4sxybs-lz4-1.8.3;/nix/store/90ba2awivci6a2wzycgd65ib31irmky2-bzip2-1.0.6.0.1-dev;/nix/store/l2i9gmhz0rsi91mz4pvvvnvzkvi5wl6b-bzip2-1.0.6.0.1-bin;/nix/store/pmvnf35kg6k2p0z91ac663akazlf7wz6-bzip2-1.0.6.0.1;/nix/store/hq0kcrwhw88a1wv1s1yvlvb1x6zcnp02-pluginlib-2.1.1;/nix/store/gahr6lk1nv1h4h6ydpwbiyfarkv29m2c-class_loader-0.4.1;/nix/store/afm6qp6xzmbxphiz4zq1yr64xpgafm0p-poco-1.9.0;/nix/store/wp4cdrk8ay9j5393b64i1216fczwaija-gpgme-1.12.0-dev;/nix/store/xd6kz4ixgzklh6v11f5ma2pz39w3lcqf-libgpg-error-1.32-dev;/nix/store/m1493r87yvgqbfsp7zgx3dr4w8z1qjv0-libgpg-error-1.32;/nix/store/xql1swlhkw4gry13vk1p0jvcaq48bmdy-glib-2.56.0-dev;/nix/store/hy22pdan86yb1mqjrhafw1fxq0ipf1sj-zlib-1.2.11-dev;/nix/store/di0d17idlw6fxf3qwpfyxciw94ywi664-zlib-1.2.11;/nix/store/07j6as6vmp31ncc5q49hqiq67fgn7bha-libffi-3.2.1-dev;/nix/store/qq9g14g39v7k1982z4512v3f13bjhg9g-libffi-3.2.1;/nix/store/vfrwql68wspvhvgrl4q7i6s1h
dp5qs0-gettext-0.19.8.1;/nix/store/89ab5vx98sf7zlnp4n5j0bwdyk50ah37-glibc-iconv-2.27;/nix/store/njs4pvn9gs1zhikh0cdn7zw4aplpmbqm-glib-2.56.0;/nix/store/fn8j5k9j0ji9zhrmnii5s4iqjxh72vx5-libassuan-2.5.1-dev;/nix/store/wccyjngxjqbpjmn7qcq3yfpqwhj6z7f0-libassuan-2.5.1;/nix/store/bw6a7smxwpacsvscbyd61q2y5sxvcfh8-pth-2.0.7;/nix/store/yb678m6r4rl2vbaqyk98q4jqcfm2qxv3-gpgme-1.12.0;/nix/store/lzgiv07pap4bvnbwh2viw3rpdc2wh18j-openssl-1.0.2p-dev;/nix/store/qvsgzbf8h2b5fxrg80mzlvj59i28xlny-openssl-1.0.2p-bin;/nix/store/ra2kcrzvp2hmsvmq82kma077h19yx0hr-openssl-1.0.2p;/nix/store/9hvcnn52byicb6qcpfrngggq358bsmpl-std_srvs-1.11.2;/nix/store/1wjrlnhld90lf0f3yqzlqyla4aidll7d-topic_tools-1.13.6;/nix/store/xhm677dqa1zvqavx1fk758qqglrfk3s4-python3.6-python-gnupg-0.4.3;/nix/store/x5ic0y8dkka3nxljc2lmhgxq64pgpa34-gnupg1compat-2.2.10;/nix/store/g9hmayip0gm88l1p8sz19ns93dl0izmf-rosmsg-1.14.4;/nix/store/77cqhxh8s8a4vd4ajrb2k09y3qymhlmz-rosnode-1.14.4;/nix/store/ghbx3pszkixd7swd0z9zhpc10dzwnia6-rostopic-1.14.4;/nix/store/binydxydc3pv88zzdxc66vc4186szrck-rosservice-1.14.4;/nix/store/sks4cvkr1rs3aq8648dfl09rgy5fz5n5-roswtf-1.14.4;/nix/store/08nn1gsmxnwzwb4q17ld6lipn3dnjkw6-actionlib-1.11.14;/nix/store/9ramhacdywjr0jm8bryryds0dmp0j695-actionlib_msgs-1.12.5;/nix/store/nizvixr8f48y58as923b5zzzxsds4hrq-ros_opcua_communication-0.2.0;/nix/store/zxgz84f9k9g6fkf43xhdmbw18yc9ba5g-ros_opcua_msgs-0.2.0;/nix/store/6fhr5b64ncw200ylsbynpxx36c14v0jd-ros_opcua_srvs-0.2.0;/nix/store/bdchcn4bsif7l1sc68mmss7wfyyk7y80-ros_opcua_impl_freeopcua-0.2.0;/nix/store/mdz6nyi8kbyqfgypah1h4n8sg9f7ghq2-freeopcua_2018-01-26;/nix/store/vhr9bh3z5zj4qgprdnpv29wyq0xsq20s-libxml2-2.9.8-dev;/nix/store/lk063qlc0pnrk9yq6sdnqw4k4rv8j752-hook;/nix/store/72mxkk74cv266snkjpz1kwl1i2rg8rpc-libxml2-2.9.8;/nix/store/h73hfs1kkr9r8zr8rgjzivdc2ff0azdq-libxml2-2.9.8-bin;/nix/store/mfghsbysp6ikwzxqf0mvc1hg86x29r2p-libxml2-2.9.8-py;/nix/store/sbg7j9ghnhl3kiw5gck9khhr2wwd5y8f-pkg-config-0.29.2;/nix/store/hq87gnfcfw6fp0lhip4sw4v53bf393va-python3.6-ipfsapi-0.4.3;/nix/store/lpl6y173ggcfd6736yxd8mb4kmn0mrz7-python3.6-requests-2.19.1-dev;/nix/store/w3lvlslna3rqwgv40m2336v6fswlvg1d-python3.6-urllib3-1.23;/nix/store/5hvsmy1bmkw10n [... cut ...]
[00:54][aira-05][root][~]# . ~/.nix-profile/setup.bash
[00:54][aira-05][root][~]# . ~/catkin_ws/devel/setup.bash
[00:54][aira-05][root][~]# roslaunch robonomics_liability liability.launch
RLException: [liability.launch] is neither a launch file in package [robonomics_liability] nor is [robonomics_liability] a launch file name
The traceback for the exception was written to the log file
[00:55][aira-05][root][~]# roscd robonomics_liability
roscd: No such package/stack 'robonomics_liability'
```
</details>

khssnv, 2018-12-01:
The same for any package build in my `catkin_ws` currenlty, maybe it's a local issue.

akru, 2018-12-26:
Should be fixed after https://github.com/airalab/airapkgs/commit/5d9c60cb0f2993e7fbf7890dceea5bd7394791ae

closed 2018-12-26 by akru