airalab/robonomics_comm #86: ROS API for IPFS

khssnv, 2018-12-24:
ROS wrapper for IPFS API? Service to `add` by given file path and response with hash in response, for example. Is it make sense?

**Pros**
* Consistence of Robonomics platfrom API, AIRA closer to sufficient ROS API for a Web3.0,
* No need to learn `ipfsapi` interface,
* No need to create and keed another [instance](https://github.com/airalab/robonomics_comm/blob/73f295c6e5ebb208cf5f5006caae52e476dea782/robonomics_liability/src/robonomics_liability/executor.py#L35) of `ipfs_client`, reimplement interraction and errors handling.

**Cons**
* One more intermediate (user -> wrapper -> ipfsapi module -> IPFS provider).

strdn, 2019-02-19:
@khssnv, ROS API (via ROS services) for upload/download files to/from IPFS is introduced in #102. 
Feel free to reopen this issue or create another one if more cases are needed.

closed 2019-02-19 by strdn

khssnv, 2019-02-19:
@strdn, great! Any dependencies on Robonomics platform specific

khssnv, 2019-02-19:
@strdn, great! Any dependencies on Robonomics platform specific components? Can we introduce it as a general IPFS ROS API package for ROS and IPFS communities?

strdn, 2019-02-22:
@khssnv, this functional implemented in ipfs_common package and can be used without robonomics_comm. But I'm think that sources of ipfs_common package must be separated from robonomics_comm and some another things must be done before introducing for communities.