airalab/robonomics_comm #88: Arithmetics for UInt256 type objects with common integers

khssnv, 2018-12-26:
Since every number in robonoimcs ROS [messages](https://github.com/airalab/robonomics_comm/blob/b810246e49ef5412004910e1adb44fd4a55f6f09/robonomics_msgs/msg/Demand.msg#L8) is [UInt256](https://github.com/airalab/robonomics_comm/blob/master/ethereum_common/msg/UInt256.msg) type user have to convert UInt256 string member to integer and vice versa to work with it. Is it possible to implement operators overloading to have arithmetics with UInt256 type objects and built-in integers?

khssnv, 2018-12-26:
Exactly. Implicit conversion on assignment built-in integer to UInt256 type field would be perfect as well.
```
>>> offer = Offer()
>>> offer.cost = 42
>>> type(offer.cost)
<class 'ethereum_common.msg._UInt256.UInt256'>
```

strdn, 2019-01-21:
@khssnv, please check. This behavior available since commit https://github.com/airalab/robonomics_comm/commit/c9c830d4c1dce3950441d77fef448629f48fc709 (currently in robonomics_comm-nightly) 

``` python
>>> from ethereum_common.msg import UInt256
>>> u1 = UInt256("123")
>>> u1 + 1
uint256: "124"
>>> u1 += 2
>>> u1
uint256: "125"
>>> 
>>> u2 = UInt256("25")
>>> u3 = u1 + u2
>>> u3
uint256: "150"
>>> type(u3)
<class 'ethereum_common.msg._UInt256.UInt256'>
>>> u4 = u3 + 1
>>> type(u4)
<class 'ethereum_common.msg._UInt256.UInt256'>
>>> 
>>> 
>>> 
>>> from robonomics_msgs.msg import Offer
>>> o = Offer()
>>> o.cost = u4
>>> o.cost
uint256: "151"
>>> o.cost += 2
>>> o.cost
uint256: "153"
>>> u5 = UInt256("7")
>>> o.cost += u5
>>> o.cost
uint256: "160"
>>> type(o.cost)
<class 'ethereum_common.msg._UInt256.UInt256'>
```

closed 2019-01-21 by strdn