airalab/robonomics_comm #103: Static objective composer / decomposer

khssnv, 2019-04-19:
In my work I've find three common types of `objective`:
1. `Static` objective with parameters necessary to have to start a proccess, like agreement values (process duration, links, names and etc.),
2. `Dynamic` objective with time sensitive data, like *cmd_vel* or mission *waypoints*,
3. `Mixed` with both types of data.

For `static` objective it's not handy to store one-shot data in pubsub topics by it's raw API, and worse to get them from topics. There is a need for helper functions to store data in static objective and get them from it by some convenient API like from dictionary, for example.

akru, 2019-04-19:
[This](https://github.com/airalab/robonomics-winter-school-2018/blob/master/fuji_weather/scripts/worker_node#L146) possible help you.

khssnv, 2019-04-19:
Closed by #103.

closed 2019-04-19 by khssnv