airalab/robonomics_comm #108: Execute arbitrary liability by address

khssnv, 2019-05-20:
Hi.

Currently one can execute liability cathing it from `liability` node API or publishing *Liability* type message in topic `/liability/incoming` with all parameters of liability contract exists in blockchian. It's possible to fetch all message fields by given liability contract address. Can you implement this API method? For example, `/liability/resume` service can be overloaded with this functionality.

strdn, 2019-05-20:
`/liability/reproduce` looks better in my opinion.

akru, 2019-05-21:
IMHO we just have a service to execute any liability by address, it also could be used by liability engine or any another staff.

strdn, 2019-08-31:
/liability/start can load and play arbitrary liability

closed 2019-08-31 by akru