airalab/robonomics_comm #111: Signer fails to use infura rinkeby testnet provider

khssnv, 2019-05-26:
Please find signer node log below.

<details>
  <summary>cat ~/.ros/log/latest/liability-infochan-eth-signer-4.log</summary>

```console
[rospy.client][INFO] 2019-05-26 16:43:36,953: init_node, name[/liability/infochan/eth/signer], pid[55163]
[xmlrpc][INFO] 2019-05-26 16:43:36,953: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2019-05-26 16:43:36,961: Started XML-RPC server [http://iotipciAira:33251/]
[rospy.init][INFO] 2019-05-26 16:43:36,961: ROS Slave URI: [http://iotipciAira:33251/]
[rospy.impl.masterslave][INFO] 2019-05-26 16:43:36,961: _ready: http://iotipciAira:33251/
[rospy.registration][INFO] 2019-05-26 16:43:36,962: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2019-05-26 16:43:36,962: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2019-05-26 16:43:37,061: registered with master
[rospy.rosout][INFO] 2019-05-26 16:43:37,062: initializing /rosout core topic
[rospy.rosout][INFO] 2019-05-26 16:43:37,068: connected to core topic /rosout
[rospy.simtime][INFO] 2019-05-26 16:43:37,074: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rospy.internal][INFO] 2019-05-26 16:43:37,287: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2019-05-26 16:43:39,770: topic[/liability/infochan/eth/signing/result] adding connection to [http://iotipciAira:34325/], count 0
[rospy.internal][INFO] 2019-05-26 16:43:39,818: topic[/liability/infochan/eth/sending/demand] adding connection to [/liability/infochan/ipfs_channel], count 0
[rospy.internal][INFO] 2019-05-26 16:43:39,819: topic[/liability/infochan/eth/sending/offer] adding connection to [/liability/infochan/ipfs_channel], count 0
[rospy.internal][INFO] 2019-05-26 16:43:39,820: topic[/liability/infochan/eth/sending/result] adding connection to [/liability/infochan/ipfs_channel], count 0
[rospy.internal][INFO] 2019-05-26 16:48:07,856: topic[/liability/infochan/eth/sending/offer] adding connection to [/rostopic_62361_1558874887707], count 1
[rospy.internal][INFO] 2019-05-26 16:49:11,160: topic[/liability/infochan/eth/signing/demand] adding connection to [http://iotipciAira:43139/], count 0
[rospy.internal][INFO] 2019-05-26 16:49:11,163: topic[/liability/infochan/eth/signing/offer] adding connection to [http://iotipciAira:43139/], count 0
[rospy.internal][ERROR] 2019-05-26 16:49:23,663: Unable to initiate TCP/IP socket to iotipciAira:42483 (http://iotipciAira:43139/): Traceback (most recent call last):
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 558, in connect
    self.read_header()
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 653, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 596, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
rospy.exceptions.TransportInitError: remote error reported: node shutting down

[rospy.internal][ERROR] 2019-05-26 16:49:23,666: Unable to initiate TCP/IP socket to iotipciAira:42483 (http://iotipciAira:43139/): Traceback (most recent call last):
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 558, in connect
    self.read_header()
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 653, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 596, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
rospy.exceptions.TransportInitError: remote error reported: node shutting down

[rospy.internal][WARNING] 2019-05-26 16:49:24,665: Unknown error initiating TCP/IP socket to iotipciAira:42483 (http://iotipciAira:43139/): Traceback (most recent call last):
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 556, in connect
    self.socket.connect((dest_addr, dest_port))
ConnectionRefusedError: [Errno 111] Connection refused

[rospy.internal][INFO] 2019-05-26 16:49:24,665: topic[/liability/infochan/eth/signing/demand] removing connection to http://iotipciAira:43139/
[rospy.internal][WARNING] 2019-05-26 16:49:24,667: Unknown error initiating TCP/IP socket to iotipciAira:42483 (http://iotipciAira:43139/): Traceback (most recent call last):
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 556, in connect
    self.socket.connect((dest_addr, dest_port))
ConnectionRefusedError: [Errno 111] Connection refused

[rospy.internal][INFO] 2019-05-26 16:49:24,667: topic[/liability/infochan/eth/signing/offer] removing connection to http://iotipciAira:43139/
[rospy.internal][INFO] 2019-05-26 16:52:04,929: topic[/liability/infochan/eth/signing/demand] adding connection to [http://iotipciAira:36735/], count 0
[rospy.internal][INFO] 2019-05-26 16:52:04,932: topic[/liability/infochan/eth/signing/offer] adding connection to [http://iotipciAira:36735/], count 0
[rosout][ERROR] 2019-05-26 16:52:15,464: Failed to get nonce by address 0xE59E92C7fD8494b926C8957359a687d094620e84 with exception: The field extraData is 97 bytes, but should be 32. It is quite likely that you are connected to a POA chain. Refer http://web3py.readthedocs.io/en/stable/middleware.html#geth-style-proof-of-authority for more details. The full extraData is: HexBytes('0xd683010900846765746886676f312e3132856c696e7578000000000000000000166baec917c27b156ef74fd648a5b200751bcd9d005108d4396af940141b9d291c119d9044bf4c77d36c642667889f3f57259c2900be18c9b36f4711f74c6ca301')
[rosout][ERROR] 2019-05-26 16:52:15,468: bad callback: <function Signer.__init__.<locals>.sign_offer at 0x7f714b187d08>
Traceback (most recent call last):
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/nix/store/7pndab5s8qsambv0brvp3y9f0nlzfs5g-robonomics_comm-4a921c44/lib/python3.7/site-packages/ethereum_common/signer.py", line 93, in sign_offer
    message_hash = robonomicsMessageUtils.offer_hash(msg, current_nonce)
  File "/nix/store/7pndab5s8qsambv0brvp3y9f0nlzfs5g-robonomics_comm-4a921c44/lib/python3.7/site-packages/robonomics_msgs/robonomicsMessageUtils.py", line 46, in offer_hash
    return Web3.soliditySha3(types, [multihash.decode(msg.model.multihash.encode(), 'base58').encode(),
  File "/nix/store/w1pmsd9rfl7damcfpc6mv9ym7mrzx012-python3.7-multihash-0.8.2/lib/python3.7/site-packages/multihash/multihash.py", line 234, in decode
    "multihash length field does not match digest field length")
ValueError: multihash length field does not match digest field length

[... 72 lines cut ...]
rospy.exceptions.TransportInitError: remote error reported: node shutting down

[rospy.internal][ERROR] 2019-05-26 16:53:06,854: Unable to initiate TCP/IP socket to iotipciAira:44177 (http://iotipciAira:36735/): Traceback (most recent call last):
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 558, in connect
    self.read_header()
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 653, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 596, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
rospy.exceptions.TransportInitError: remote error reported: node shutting down

[rospy.internal][WARNING] 2019-05-26 16:53:07,854: Unknown error initiating TCP/IP socket to iotipciAira:44177 (http://iotipciAira:36735/): Traceback (most recent call last):
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 556, in connect
[... cut ...]
[rospy.internal][WARNING] 2019-05-26 16:53:07,855: Unknown error initiating TCP/IP socket to iotipciAira:44177 (http://iotipciAira:36735/): Traceback (most recent call last):
  File "/nix/store/7nqfy4nck72grhii35kw2pkqvya04x93-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 556, in connect
    self.socket.connect((dest_addr, dest_port))
ConnectionRefusedError: [Errno 111] Connection refused

[rospy.internal][INFO] 2019-05-26 16:53:07,855: topic[/liability/infochan/eth/signing/offer] removing connection to http://iotipciAira:36735/
```
</details>

akru, 2019-06-07:
@strdn https://web3py.readthedocs.io/en/stable/middleware.html#geth-style-proof-of-authority could help to fix it.

SyedMuhamadYasir, 2019-08-30:
> @strdn https://web3py.readthedocs.io/en/stable/middleware.html#geth-style-proof-of-authority could help to fix it.

Hey there, i have a similar problem.

while trying to connect to publish a smart contract to Rinkeby ( while using Web3.py, Infura and Remix ) i get the extra data error

i don't know how to deploy this middle-ware WITHOUT using geth. is it possible to avoid using geth ?

akru, 2019-08-30:
Hello @SyedMuhamadYasir, unfortunately no, Rinkeby is geth-only network. But web3py has middleware hook to work with it.

SyedMuhamadYasir, 2019-08-30:
> Hello @SyedMuhamadYasir, unfortunately no, Rinkeby is geth-only network. But web3py has middleware hook to work with it.

i see.

can you please refer a step by step guide to me, which could help me hook the middleware to the web3.py ?

I am actually quite confused on how to approach the issue of connecting web3.py to Rinkeby.

i have installed Geth but i am at a loss on how to use the Rinkeby network through geth, while using web3.py. ( i'm not sure my machine can handle the load of being a full node )

any help or guidance you can provide is much appreiciated @akru !

akru, 2019-09-01:
@SyedMuhamadYasir, you can see how to use this hook in project sources:

https://github.com/airalab/robonomics_comm/blob/master/robonomics_liability/src/robonomics_liability/listener.py#L41

Discussion is off topic, close.

closed 2019-09-01 by akru