airalab/robonomics_comm #114: Warning /liability/executor is not a publisher of ...

Vourhey, 2019-06-03:
robonomics_comm 1.0.0
airalab-sidechain

I see the following log 
```
[rosout][WARNING] 2019-06-03 10:10:26,866: Could not process inbound connection: [/liability/executor] is not a publisher of [/liability/eth_0x5820106C0B44C51150aD04eFD0c16a4ce851d09A/task]. Topics are [['/rosout', 'rosgraph_msgs/Log'], ['/liability/persistence/add', 'robonomics_liability/Liability'], ['/liability/persistence/del', 'robonomics_liability/Liability'], ['/liability/complete', 'robonomics_liability/Liability'], ['/liability/ready', 'robonomics_liability/Liability'], ['/liability/result', 'robonomics_msgs/Result'], ['/liability/persistence/update_timestamp', 'robonomics_liability/LiabilityExecutionTimestamp']]{'callerid': '/worker', 'md5sum': '992ce8a1687cec8c8bd883ec73ca41d1', 'message_definition': 'string data\n', 'tcp_nodelay': '0', 'topic': '/liability/eth_0x5820106C0B44C51150aD04eFD0c16a4ce851d09A/task', 'type': 'std_msgs/String'} 
```
this message repeats again and again. Validator is 0x0...0

akru, 2019-06-04:
Could you provide steps to reproduce this error.

Vourhey, 2019-06-11:
I tried several agents. It doesn't really affect the process, but warning messages don't stop.

The steps are usual:
* Get a liability
* call /liability/start service
* complete the task
* call /liability/finish service

In a dapp I can see the result and it's correct. I mean I'm able to download the result Qm... file and play it with the rosbag tool.

Vourhey, 2019-08-07:
On the agent side it shows as an error
```
[rospy.internal][ERROR] 2019-08-07 13:47:35,246: Unable to initiate TCP/IP socket to dczd-aira-droneid:36577 (http://dczd-aira-droneid:38673/): Traceback (most recent call last):
  File "/nix/store/34kbwpb98973b7f4w83scg2g5v2dq0h0-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 558, in connect
    self.read_header()
  File "/nix/store/34kbwpb98973b7f4w83scg2g5v2dq0h0-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 653, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
  File "/nix/store/34kbwpb98973b7f4w83scg2g5v2dq0h0-rospy-1.14.4/lib/python3.7/site-packages/rospy/impl/tcpros_base.py", line 596, in _validate_header
    raise TransportInitError("remote error reported: %s"%header['error'])
rospy.exceptions.TransportInitError: remote error reported: [/liability/executor] is not a publisher of [/liability/eth_0x9Ea09A69c8F46c7E45C4eD01866010B191614f03/pilot_reg]. Topics are [['/rosout', 'rosgraph_msgs/Log'], ['/liability/persi
stence/add', 'robonomics_liability/Liability'], ['/liability/persistence/del', 'robonomics_liability/Liability'], ['/liability/complete', 'robonomics_liability/Liability'], ['/liability/ready', 'robonomics_liability/Liability'], ['/liabili
ty/result', 'robonomics_msgs/Result'], ['/liability/persistence/update_timestamp', 'robonomics_liability/LiabilityExecutionTimestamp'], ['/liability/eth_0xA1c3da919d071135C8862e6aDd87Bf33723F9f37/id_serial', 'std_msgs/String'], ['/liabilit
y/eth_0xA1c3da919d071135C8862e6aDd87Bf33723F9f37/email', 'std_msgs/String'], ['/liability/eth_0xA1c3da919d071135C8862e6aDd87Bf33723F9f37/drone_type', 'std_msgs/String'], ['/liability/eth_0xA1c3da919d071135C8862e6aDd87Bf33723F9f37/drone_mak
e', 'std_msgs/String'], ['/liability/eth_0xA1c3da919d071135C8862e6aDd87Bf33723F9f37/drone_model', 'std_msgs/String'], ['/liability/eth_0xA1c3da919d071135C8862e6aDd87Bf33723F9f37/id_reg', 'std_msgs/String'], ['/liability/eth_0xA1c3da919d071
135C8862e6aDd87Bf33723F9f37/pilot_name', 'std_msgs/String'], ['/liability/eth_0xA1c3da919d071135C8862e6aDd87Bf33723F9f37/pilot_reg', 'std_msgs/String']]

```

Vourhey, 2019-08-19:
I think this might be an issue due to this unstopable warnings/errors the next liability is not executed

akru, 2019-08-20:
It doesn’t looks like robonomics_comm problem. Could you reallocate subscriber on agent side when liability finished?

Vourhey, 2019-08-20:
It looks like it. I added `unregister()` to every publisher/subscriber and I don't see those warnings. But it should be tested more

akru, 2019-08-22:
Great, please test and close it.

strdn, 2019-08-31:
I reproduced this problem.
Error messages occur if after the rosbag player stops (call /liability/finish), there are subscribers to the topics of the finished liability.
It's doesn't look like robonomics_comm problem.

closed 2019-09-01 by akru