airalab/robonomics_comm #116: The ipfsapi library is deprecated

Vourhey, 2019-06-25:
I built robonomics_comm on a fresh Ubuntu 18.04. After launching liability and erc20 nodes I got warnings:
```
/ws/src/robonomics_comm/ipfs_common/src/ipfs_common/ipfs_channel.py:13: FutureWarning: The `ipfsapi` library is deprecated and will stop receiving updates on the 31.12.2019! If you are on Python 3.5+ please enable
 and fix all Python deprecation warnings (CPython flag `-Wd`) and switch to the new `ipfshttpclient` library name. Python 2.7 and 3.4 will not be supported by the new library, so please upgrade.                                            
  import ipfsapi
/.local/lib/python3.6/site-packages/ipfsapi/client/base.py:29: FutureWarning: IPFS API function “pubsub_subscribe” has been renamed to “pubsub.subscribe”                                                                          
  warnings.warn(message, FutureWarning)
```
And from erc20 node:
```
/.local/lib/python3.6/site-packages/ipfsapi/client/base.py:29: FutureWarning: IPFS API function “pubsub_peers” has been renamed to “pubsub.peers”                                                                                  
  warnings.warn(message, FutureWarning)
pubsub <ipfsapi.client.Client object at 0x7f9e689b8780>
peers <ipfshttpclient.client.pubsub.Section object at 0x7f9e689b8668>
/.local/lib/python3.6/site-packages/ipfsapi/client/base.py:29: FutureWarning: IPFS API function “name_publish” has been renamed to “name.publish”                                                                                  
  warnings.warn(message, FutureWarning)
name <ipfsapi.client.Client object at 0x7f9e689b8780>
publish <ipfshttpclient.client.name.Section object at 0x7f9e6810bf28>
```
I think it's time to switch to ipfshttpclient library

akru, 2019-09-01:
Merged

strdn, 2019-10-01:
https://github.com/ipfs/py-ipfs-http-client/issues/193

strdn, 2019-10-01:
https://github.com/airalab/robonomics_comm/issues/130

closed 2019-11-12 by akru