airalab/robonomics_comm #121: Proposal: change start/finish liability service type to Liability

Vourhey, 2019-08-20:
Every time to start a liability I have a code like this:
```
def __init__(self):
    ...
    rospy.wait_for_service("/liability/start")
    self.liability_start = rospy.ServiceProxy('/liability/start', StartLiability)
    ...

def on_liability(self, liability):
    ...
    self.liability_start(liability.address.address)
```
and similar to call finish liability

I think it would be better to change the type of StartLiability from:
```
string address
---
bool success
string msg
```
to
```
Liability liability
---
bool success
string msg
``` 
to make call easier to read and look better like:
```
self.liability_start(liability)
```
The same is for finish service

akru, 2019-08-22:
In my opinion pass full liability structure to this call is more than needed. Liability ID like an address is enough.

strdn, 2019-08-31:
I'm agree with @akru. Address is necessary and sufficient.

strdn, 2019-09-01:
Change from
```
string address
---
bool success
string msg
```
to 
```
Address address
---
bool success
string msg
```
is needed

closed 2019-09-01 by strdn